jordanebelanger/SwiftyBluetooth #2: Xcode 8 integration

szpako1, 2016-09-27:
Can't compile under Xcode 8. Conversion to swift 2.3 or 3.0 is needed.

tehjord, 2016-09-27:
50 errors anddddd dropping :) Swift 3 a real pain in the butt. 

I'll have to make the swift 3 branch the v2.0 of the library once it's ready, so let me know if there are any other fix you would like to see while we're at it.

szpako1, 2016-09-28:
@tehjord no other fixes are needed. Thanks for your job.

szpako1, 2016-10-24:
@tehjord hi, how about it?

tehjord, 2016-10-25:
@szpako1 Hi, currently working on it on and off, Swift3 notably require to do a pretty significant refactor of the internal Error enum, shouldnt take much longer hopefully.

tehjord, 2016-11-03:
@szpako1 Just released the first draft of the swift3 version, keep watching the swift3 branch for updates as I test it out and iron out bugs that might have been created during conversion.

szpako1, 2016-11-28:
@tehjord Hi, I can't use that in my podspec file as dependency because podspec didn't have support for selecting branch / git path :|

When do you plan to marge swift3.0 branch to master?

tehjord, 2016-11-28:
Ah sorry about that, I didn't know you couldn't do that with Cocoapods, I've been using Carthage for sometimes now. Let me look into merging this thing today if I have time.

tehjord, 2016-11-28:
@szpako1 Alright I pushed the Swift 3 branch to master and updated the documentation + podspec repo. Enjoy!

closed 2016-12-19 by tehjord