jordanebelanger/SwiftyBluetooth #12: Not able to notify on characteristic

jakerockland, 2016-12-13:
Hey @tehjord, hoping you can shine some light on a bug I've been struggling with today. I've been testing with your fork rather than ours. With the source that is currently on `master` I am able to successfully set a characteristic to notify with the `Peripheral.setNotifyValue()` method. However, with the source on the `result` branch, I no longer am able to.

If I attempt to multiple times, the first time I do I get this error: `Bluetooth operation timed out: Discover services`

The second time I do I get this error: `Bluetooth operation timed out: Discover characteristics`

And then any additional time I try after that I get this error: `Bluetooth operation timed out: Update notification status`

Would really appreciate any thoughts you might have as to what might be causing the problem, would like to be using the changes on the `result` tree but need to be able to notify on characteristics 😓

jakerockland, 2016-12-13:
I think I was actually mistaken here. Not sure exactly why I'm experiencing this, but I think it is most likely due to something I'm doing, not anything with the library here. Closing this for now, sorry about that 😅

closed 2016-12-13 by jakerockland

tehjord, 2016-12-13:
Hey know worries man, thanks for testing the branch.