jordanebelanger/SwiftyBluetooth #19: Bluetooth operation timed out: Discover services

AnnieNinaJoyceV, 2017-02-18:
Hello! I would like to know why I am getting this error. This is my first attempt in connecting to a BLE device. Here is the code I am using,

[image]

Am I setting any 'Timeout', I don't see any such in my code. Thanks in advance.

jakerockland, 2017-02-22:
@AnnieNinaJoyceV My guess is that you have some sort of mismatch in the UUID that you are using to try to discover services on the Bluetooth device you're connected to and the actual UUIDs of those services on the device. Have you used any tools that allow you to explore the UUIDs of services and characteristics on the device you're trying to connect with so that you can verify you're using the correct values?

jakerockland, 2017-02-22:
Nordic Semiconductor makes a useful iOS app that lets you do this, its called [nRF Connect](https://itunes.apple.com/us/app/nrf-connect/id1054362403?mt=8).

jakerockland, 2017-02-22:
Additionally, the timeout behavior is a default behavior (you don't have to explicitly set a timeout) that is set up to give a response when no service or characteristic is found within the expected amount of time usually needed.

AnnieNinaJoyceV, 2017-02-23:
@jakerockland Yes, I am using valid UUIDs. Surprisingly it works if I use plain CoreBluetooth library. Not sure why it's not working when I use SwiftyBluetooth library.

closed 2017-09-24 by jordanebelanger