jordanebelanger/SwiftyBluetooth #21: Connection timeout timer does not fire

agildehaus, 2017-06-20:
I'm unable to get a reliable connection timeout from this library.  I've looked into it briefly and it seems the timer might be going out of scope?

I've found for these timers, Grand Central Dispatch makes cleaner code (no selectors, no @objc, no weak vars).  Try doing something like this instead of using the Timer class:

```
// CentralProxy class-level

var connectTimer: DispatchSourceTimer?

// connect(peripheral:timeout:_:)

self.connectTimer = DispatchSource.makeTimerSource()
self.connectTimer!.setEventHandler {
    let uuid = request.peripheral.identifier

    self.connectRequests[uuid] = nil

    self.centralManager.cancelPeripheralConnection(request.peripheral)

    request.invokeCallbacks(error: SBError.operationTimedOut(operation: .connectPeripheral))
}

self.connectTimer!.scheduleOneshot(deadline: .now() + timeout, leeway: .seconds(0))
self.connectTimer!.resume()
```

jordanebelanger, 2017-06-20:
Once you have scheduled a timer on a runloop, your reference to that timer might go out of scope if you don't retain it somewhere, but the actual timer is still present on your runloop until it is either validated or it runs its course and calls its selector.  

Might I see your connect code?

There is no connection timeout by default, i.e 

peripheral.connect { ... }
or 
peripheral(withTimeout: nil) { ... }

will never timeout, is that perhaps what you are doing?

agildehaus, 2017-06-20:
```
func connect(timeout: TimeInterval = 10.0) -> Promise<Void> {
        return Promise { resolve, reject in
            print("connect()")
            self.peripheral.connect(withTimeout: timeout, completion: { result in
                switch result {
                case .failure(let error):
                    reject(error)
                case .success:
                    self.peripheral.setNotifyValue(
                        toEnabled: true,
                        forCharacWithUUID: GKCard.controlPointUUID,
                        ofServiceWithUUID: GKCard.serviceUUID
                    ) { result in
                        switch result {
                        case .failure(let error):
                            let nsError = error as NSError
                            switch nsError.code {
                            case CBATTError.insufficientEncryption.rawValue:
                                reject(CardError.cardNotPaired)
                            default:
                                reject(error)
                            }
                        case .success:
                            print("connected")
                            resolve()
                        }
                    }
                }
            })
        }
    }
```

I'm using [Hydra](https://github.com/malcommac/Hydra), a Swift promises library.

Everything works great except for the timeout.  I modified the timer as I described above which fixes it. I'll do a bit more investigation into the cause.

jordanebelanger, 2017-06-20:
Humm that is a bit disconcerting, lemme test the connect timeout stuff and see for myself. 

What about threading, are the promises executing on the main thread?

agildehaus, 2017-06-20:
Hydra runs the promise body on it's own DispatchQueue ([source](https://github.com/malcommac/Hydra/blob/master/Sources/Hydra/Promise.swift#L45)).

jordanebelanger, 2017-06-20:
You can manually set the dispatch queue you want your Hydra promise to run on, try and run the promise on the mainQueue and see if your timeout works.

agildehaus, 2017-06-20:
Okay, running the connection promise on the main thread does resolve the issue.

jordanebelanger, 2017-06-20:
Shit, in that case I am not sure what's going on, is the timer not working either when you use it outside of your promise?

I don't particularly like the NSTimer synthax either, but the entire library is based around the same NSTimer pattern and timeouts seems to be working in most places as far as I know. Why you the connect timeout is having issue in your case I am not sure, I will test later on my side to see if I can repro.

agildehaus, 2017-06-20:
I think you misread my last comment? :) . The issue goes away when running on the main thread.

jordanebelanger, 2017-06-20:
Woops, I read that as "does not" instead of "does" hehe

Glad to know it's working, enjoy!

closed 2017-06-22 by jordanebelanger