jordanebelanger/SwiftyBluetooth #22: Reading old advertisement data

Firefox2005, 2017-06-26:
Hey, 

we are using SwiftyBluetooth for communicating with our own peripheral. We are sending important data in the advertisement data (manufacturer specific data) which change frequently. Against all I was able to find in the internet my iPhone 6 cashes the advertisement data and therefore most of the time I get old data when scanning. 

I was able to fix this behavior when passing the following option in the 
`centralManager.scanForPeripherals(withServices: serviceUUIDs, options: [CBCentralManagerScanOptionAllowDuplicatesKey:1])`

This allows me to use the second package I receive which always contains the current values, while the first package still contains the old data.
Is there any way of doing that already? Is it even desired to make this possible? Or do you see a different way for achieving that?

Thanks

jordanebelanger, 2017-06-26:
Hey there,

I remember having a similar issue while working on my own bluetooth devices whereas old peripheral would be cached and not discovered again, really annoying!

Looks like I simply forgot to add that functionality in the library afterwards.

jordanebelanger, 2017-06-26:
Should be fixed in https://github.com/jordanebelanger/SwiftyBluetooth/commit/6a8ef50b268ceff79ad5dee5a4ea1a6b3c584f49

Firefox2005, 2017-06-26:
Great, thanks! This will help us a lot!

closed 2017-06-26 by Firefox2005