jordanebelanger/SwiftyBluetooth #23: no way to retrieve peripherals

tcurdt, 2017-08-21:
I don't see this exposed: 

https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1519127-retrieveperipherals

jordanebelanger, 2017-08-21:
Humm :S 

https://github.com/jordanebelanger/SwiftyBluetooth/blob/e38dfdfb7686e99d983675988bc6228705919464/SwiftyBluetooth/Source/SwiftyBluetooth.swift#L71

https://github.com/jordanebelanger/SwiftyBluetooth/blob/e38dfdfb7686e99d983675988bc6228705919464/SwiftyBluetooth/Source/Central.swift#L168

tcurdt, 2017-08-21:
Hm. Seems like it's just that the docs are out of date then:

http://cocoadocs.org/docsets/SwiftyBluetooth/0.2.0/

jordanebelanger, 2017-08-21:
Yea looks like cocoadoc is broken and won't update for the latest version, prolly because I changed my account name and something broke. 

Looks like we're done providing documentation outside of code :) removed the link from the readme

closed 2017-08-21 by jordanebelanger

jakerockland, 2017-08-22:
@jordanebelanger have you seen this? https://github.com/realm/jazzy/ 😊