jordanebelanger/SwiftyBluetooth #29: Fix warnings on simulator?

xinsight, 2018-01-31:
Hi,

When running this code in the simulator, this code warnings on the console:

```
SwiftyBluetooth.setSharedCentralInstanceWith(restoreIdentifier: "xxx")

if SwiftyBluetooth.Central.sharedInstance.state == .poweredOn {
    SwiftyBluetooth.scanForPeripherals(withServiceUUIDs: nil, options: nil, timeoutAfter: 30) { result in
        print("result: \(result)")
    }
}
```

```
[CoreBluetooth] XPC connection invalid
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x600000078a80> can only accept this command while in the powered on state
```

Is there a way to avoid the `API MISUSE` warning? I haven't been able to determine what the actual command is that is triggering the warning, but it must be getting called from within SwiftyBluetooth. 

Any ideas?

xinsight, 2018-01-31:
Did a bit more digging and it seems that the XPC failure `[CBXpcConnection _handleInvalid]` eventually sets the `CBCentralManager.state` to `.unsupported` which triggers a call to `-CentralProxy.stopScan` which calls `-CBCentralManager.stopScan()`. And you can't stop scanning when the state is unsupported or powered down. I'll make a PR.

jordanebelanger, 2018-01-31:
Thank you for the very nice find, will merge promptly.

closed 2018-01-31 by jordanebelanger