jordanebelanger/SwiftyBluetooth #34: Notification on natural disconnection

jeyremy, 2018-08-28:
Hi, thanks for this amazing lib. 

I was wondering, i have a project where the phone has do be connected as much as possible to my peripheral. 
This being say, i have to do connection with no-timeout, (this is working, amazing) 
But when the distance is too far, after it's already connected, the peripheral and the iPhone disconnects. at that time i have to send a re-connection request, with no timeout.

So my question, is there a way to get notified on natural disconnection so i can re-launch a connexion request ? 

In advance, thanks for your time and your amazing work.

jeyremy, 2018-10-08:
@jordanebelanger is it something you planning to do ? 
Or i can try to do it myself and submit a pull request ?  i don't really know where / when i have to do the notification but i can search. 

do you have any recommendations ? =)

jordanebelanger, 2018-10-08:
Hi Jeremy,

Thanks for your kind words and interest in the library. I have been very busy recently, but let me take a look. It would probably be just a matter of adding a new NSNotificationCenter notification being sent from the Central whenever a Peripheral disconnects.

jordanebelanger, 2020-11-19:
https://github.com/jordanebelanger/SwiftyBluetooth/releases/tag/2.2.0

closed 2020-11-19 by jordanebelanger