jordanebelanger/SwiftyBluetooth #36: I can't able to set notification.

ashokKOMARA, 2018-10-12:
failure Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}

This is the error that aim getting while adding notification to peripheral characteristic. Can u please suggest me solution.

jordanebelanger, 2018-10-12:
Can I see the code you are using for your call?

ashokKOMARA, 2018-10-12:
[image]

ashokKOMARA, 2018-10-12:
Hi, Please find bellow image. Some how I am not able to find Descriptors for the particular Characteristic, but those were there in in my peripheral. Even I tried with predefined method, but no use. it is returning empty.

[cid:EF29B244-B83E-4ADA-ABE7-BE0E016D3D2D]

[cid:223EFAA7-1902-4048-AC28-D5698EB8236C]
Regards,
Ashok Komara | iOS Developer,
Ph.no<http://Ph.no> :- +918688114115,
Email :- ashok@huginnovations.com<mailto:ashok@huginnovations.com>.

******************************************************************************************************************************************************
This e-mail is for the sole use of the intended recipient(s) and may contain confidential and privileged information. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies and the original message. Any unauthorized review, use, disclosure, dissemination, forwarding, printing or copying of this email are strictly prohibited. Do Not Print this email unless required. SAVE PAPERS. SAVE TREES. GO GREEN.
******************************************************************************************************************************************************

ashokKOMARA, 2019-01-03:
it is manufacture defect. i can able to get notification update even setnotify return failure response.

closed 2019-01-03 by ashokKOMARA