jordanebelanger/SwiftyBluetooth #37: I am getting service list nil, when i disconnect & connect again, with out killing the app.

ashokKOMARA, 2019-01-03:
I can able to get services list when i connect firs time with peripheral,then disconnected > connect again, now iam unable to list out any services.

marcospolanco, 2019-01-06:
@ashokKOMARA I am having the exact same issue. Fix?

ashokKOMARA, 2019-01-07:
No, but when i try to reconnect with same peripheral it is working fine. Some times when peripheral is connected in background it is returning the same nil.

kshrikant, 2019-04-19:
Hey anyone got the solution on this?

kshrikant, 2019-04-19:
#12

kshrikant, 2019-04-22:
Hi @ashokKOMARA , @marcospolanco @jordanebelanger Anyone know, how to resolve this  ?

Dhaval3110, 2020-02-17:
Hello @kshrikant @ashokKOMARA @marcospolanco @jordanebelanger any update on this ? #39  check this also.

jordanebelanger, 2020-02-17:
https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518706-discoverservices

closed 2020-02-17 by jordanebelanger