jordanebelanger/SwiftyBluetooth #38: Peripheral disconnect recognition

quanguyen, 2019-02-26:
Hi,

How should I detect a peripheral disconnected from the master?

Thanks,

komaxx, 2019-11-05:
Same issue here: How can I monitor an already established connection?

jordanebelanger, 2019-11-05:
> Same issue here: How can I monitor an already established connection?

Hi @komaxx, sorry to hear you're having issues, I've been getting overloaded with work, notably server side Swift stuff and haven't had much time to work on this library.

Do you know of any specific CoreBluetooth hook/notification that does what you need in this case and if so could you share a link to the documentation so I can take a look?

If it's not too hard, you may even want to consider doing a PR for it and I will gladly merge it.

adamontherun, 2020-05-31:
@jordanebelanger I have the same issue. I think the fix would be to expose the centralManager(_:didDisconnectPeripheral:error:) delegate method via your framework.

sirowain, 2020-11-19:
Any updates on this request? I'm also interested.

jordanebelanger, 2020-11-19:
@sirowain https://github.com/jordanebelanger/SwiftyBluetooth/releases/tag/2.2.0

closed 2020-11-19 by jordanebelanger

sirowain, 2020-11-20:
This is not always working, the method returns before calling the postCentralEvent if you don't have any disconnectRequest for the uuid. If you are trying to detect an unexpected peripheral disconnect you'll miss it.
```
let uuid = peripheral.identifier
guard let request = disconnectRequests[uuid] else {
    return
}
...
postCentralEvent(Central.CentralCBPeripheralDisconnected, userInfo: userInfo)
```

jordanebelanger, 2020-11-20:
> This is not always working, the method returns before calling the postCentralEvent if you don't have any disconnectRequest for the uuid. If you are trying to detect an unexpected peripheral [...]

Good catch, I made this very fast yesterday.

A pr with the fix would be appreciated.

jordanebelanger, 2020-11-20:
@sirowain Fixed in 2.2.1