jordanebelanger/SwiftyBluetooth #39: discoverCharacteristics 1 failure  -operationTimedOut(operation: SwiftyBluetooth.SBError.SBOperation.discoverServices)

kshrikant, 2019-04-18:
Hello,

I'm scanning for below CBUUIDs using `SwiftyBluetooth.scanForPeripherals` 

` let serviceId3 = CBUUID(string:"FFF0") 
   let serviceId4 = CBUUID(string:"FFB0") `

these service ids are used for detecting temperature devices.

Ill get scanned devices successfully, after that I do successful connection with those peripheral using  `selectedPeripheral!.connect(withTimeout: 10)`

I'm calling `func discoverServices`  for `serviceId = CBUUID(string:"FFE0")`

After that the issue I'm getting in your library is

 **discoverCharacteristics 1 failure  -operationTimedOut(operation: SwiftyBluetooth.SBError.SBOperation.discoverServices)**

I did deep debugging in your library so I got to know that here in below code 

    extension PeripheralProxy  {

    func discoverServices(_ serviceUUIDs: [CBUUID]?, completion: @escaping ServiceRequestCallback)  {

        self.connect { (result) in
            if let error = result.error {
                completion(.failure(error))
                return
            }
             
            if let serviceUUIDs = serviceUUIDs {
                let servicesTuple = self.cbPeripheral.servicesWithUUIDs(serviceUUIDs)
                
                if servicesTuple.missingServicesUUIDs.count == 0 {
                    completion(.success(servicesTuple.foundServices))
                    return
                }
            }
            
            let request = ServiceRequest(serviceUUIDs: serviceUUIDs) { result in
                completion(result)
            }
            
            self.serviceRequests.append(request)
            
            if self.serviceRequests.count == 1 {
                self.runServiceRequest()
            }
        }
    }

In file `PeripheralProxy.swift` on line number 183 or in above code      
           
    let servicesTuple = self.cbPeripheral.servicesWithUUIDs(serviceUUIDs)
                
     if servicesTuple.missingServicesUUIDs.count == 0 {
              completion(.success(servicesTuple.foundServices))
              return
     }

Whenever I'm getting above timeout error at that time serviceTuple.missingServicesUUIDs.count is one.

So whatever serviceId I've requested i.e `serviceId = CBUUID(string:"FFE0")` is missing. Kindly please let me know how can I resolve this, I'm trying to resolve this issue since last 3-4 days. Please help me. Thanks in advance.

**NOTE :- This is issue is not getting generated always its happening few times and sometimes serviceTuple.missingServicesUUIDs.count is zero and I'm getting success**

jordanebelanger, 2019-04-18:
Are you sure service "FFE0" exists on the peripheral you are working with? 

Tip, try to use one of the bluetooth utility app on the appstore to check, for example: https://itunes.apple.com/us/app/ble-nearby/id1069714373

Use that and you should be able to see all public services on the on your peripheral

kshrikant, 2019-04-19:
@jordanebelanger Thanks! for quick reply. As you said I've downloaded sample and tested the services are fine. The actual issue is whenever app killing and launching again app (or first time) at that time only its receiving characteristics. When within app I'm doing disconnect and cancelPeripheral connection still its not receiving characteristics. Any idea with this problem ? Or what can I do with Peripheral class objects ? Please help me thanks.