jordanebelanger/SwiftyBluetooth #47: How to get the state when the peripheral is disconnected?

mobileguru121, 2021-01-05:
func centralManager(_ central: CBCentralManager, didConnect peripheral: CBPeripheral)
I need to get this callback when the peripheral is disconnected.
How can I get this to work?
Many thanks.

mobileguru121, 2021-01-05:
I will use original CBCentralDelegate and CBPeripheralDelegate.

closed 2021-01-05 by mobileguru121

jordanebelanger, 2021-01-05:
@mobileguru121 If you want, I believe there is an NSNotification you can listen for. Check the interface of the Peripheral class it should be there.

mobileguru121, 2021-01-05:
Thank you, but I could not find it. I appreciate if you can teach me.

mobileguru121, 2021-01-05:
Ok, thank you. I have to update the pod version to latest.

mobileguru121, 2021-01-05:
Can you let me know the latest pod version?
I installed 5.0 but I can't find.