jordanebelanger/SwiftyBluetooth #48: Compilation under Xcode 13

PFRene, 2021-09-16:
I'm getting loads of warnings when trying to compile this in the latest Xcode 13 RC (with Swift5) - mostly all errors are related to this sort of thing : "Argument type 'CBService?' does not conform to expected type 'CBUUIDConvertible'"

Any thoughts or suggestions? I'm having trouble working it out

jordanebelanger, 2021-09-22:
Hello, I did a somewhat dirty fix (explicit unwrappings...) but I think it shouldn't cause any issue.

If you want, try with this branch https://github.com/jordanebelanger/SwiftyBluetooth/tree/swift5.5 until I finish merging the changes, need to make them backward compatible with Swift older than 5.5, that will take some time.

PFRene, 2021-09-28:
Thanks! The 5.5 branch will help in the mean time, at least it compiles again ;)

Rolvar, 2021-10-29:
Thanks @jordanebelanger 👏 🍻