jordanebelanger/SwiftyBluetooth #51: Xcode 13.1, v3.0.0 unhandled file warning

roberthartman, 2021-11-24:
After updating to Xcode 13.1 and SwiftBluetooth 3.0.0, included via Swift Package Manager, I now see the following warning:
[image]

```
found 1 file(s) which are unhandled; explicitly declare them as resources or exclude from the target
/Users/rob.hartman/Library/Developer/Xcode/DerivedData/iOS_Example-cvpcvvkpubwmmuebviedncjenzcp/SourcePackages/checkouts/SwiftyBluetooth/Sources/Info.plist
```

One solution might be to modify `Package.swift` targets array, exluding that file:
```targets: [
        .target(name: "SwiftyBluetooth", path: "Sources", exclude: [Sources/Info.plist"])
    ]
```

jordanebelanger, 2022-01-10:
@roberthartman Should be fixed in https://github.com/jordanebelanger/SwiftyBluetooth/releases/tag/3.0.1 thanks for reporting 👍

closed 2022-01-10 by jordanebelanger

roberthartman, 2022-01-11:
Thanks for the library!