jordanebelanger/SwiftyBluetooth #52: Expose CBCentralManager publicly

roberthartman, 2022-01-18:
I need to integrate with a legacy iOS Bluetooth library that is based upon being able to initially pass in a `CBCentralManager` instance. The library saves off the current `CBCentralManagerDelegate`, uses the central instance to do some BLE stuff, then restores the `CBCentralManagerDelegate` once it is done with its work.

I do know that I am able to get the `SwiftyBluetooth.Central` singleton via `Central.setSharedInstanceWith(restoreIdentifier:)`, and then I can use reflection to get `SwiftyBluetooth.Central.sharedInstance.centralProxy.centralManager`. This works, except that the CoreBluetooth state restoration seems to cause unwanted side effects, such as auto-connecting to my BLE peripherals, which I do not want. (I admit that I am not familiar with CoreBluetooth central state restoration... maybe there is a way to avoid things like auto-connecting?). The iOS application I am working on has pretty strict requirements for how it interacts with BLE peripherals.

So it seems like my next option is to fork SwiftyBluetooth and try to find a not-too-kludgey way of exposing `SwiftyBluetooth.Central.sharedInstance.centralProxy.centralManager` publicly.

Does this seem like a reasonable feature request?

jordanebelanger, 2022-01-18:
It wouldn't be that crazy to have a new `setSharedInstance` static function that accepts a `CBCentralManager` instance as a parameter, but most of the time the library hides all the Corebluetooth stuff from you to avoid undesired side effects/concurrency issues. 

But this is an old library that I don't have much time with, quite frankly I'd rewrite the entire thing since this is just a quick library I created back when I was learning Swift coming in from Obj-C. At this point I am mostly just keeping things up to date in terms of building/package management/swift version out of courtesy for the existing users, but not really planning on adding any new features. 

As such, I'd recommend you just go ahead and make your own fork, or give me a PR with `Central.setSharedInstance` static function that accepts a `CBCentral` instance and I'll be happy to merge it :)

roberthartman, 2022-01-18:
Thank you Jordan. I just realized that Central.sharedInstance is public! I don't know how I didn't notice that before. So I think I can use sharedInstance directly, then use reflection to get centralManager as I described to get me out of my immediate bind, but I will keep in mind what you suggest, passing in a CBCentralManager.

jordanebelanger, 2022-01-18:
> Thank you Jordan. I just realized that Central.sharedInstance is public! I don't know how I didn't notice that before. So I think I can use sharedInstance directly, then use reflection to get [...]

Seems like its your lucky day ;)

closed 2022-01-18 by jordanebelanger