jordanebelanger/SwiftyBluetooth #60: Discover services issue in 16.4 iOS

bankiprudhvi, 2023-03-31:
I have updated iOS 16.4 

I got the log below:
Error while reading #######:Bluetooth operation timed out: Discover services

Note : below 16.4 working fine

Neha-Zco, 2023-04-13:
I'm experiencing the same issue with iOS 16.4 and later versions.
Have you been able to find a solution for this issue?

jordanebelanger, 2023-04-13:
Whats the code you are using for searching, what services are you looking for, did you try looking for other services? Does it times out if you don't pass any service UUIDs when searching?

A lil more context would be appreciated...

Neha-Zco, 2023-04-13:
I have tried both ways (with service UUIDs and nil), and it still gets timed out. 

The app was using the below code.

`    fileprivate func runServiceRequest() {
        guard let request = self.serviceRequests.first else {
            return
        }
        
        self.cbPeripheral.discoverServices(request.serviceUUIDs)
        
        Timer.scheduledTimer(
            timeInterval: PeripheralProxy.defaultTimeoutInS,
            target: self,
            selector: #selector(self.onServiceRequestTimerTick),
            userInfo: Weak(value: request),
            repeats: false)
    }
    
    @objc fileprivate func onServiceRequestTimerTick(_ timer: Timer) {
        defer {
            if timer.isValid { timer.invalidate() }
        }
        
        let weakRequest = timer.userInfo as! Weak<ServiceRequest>
        
        // If the original rssi read operation callback is still there, this should mean the operation went
        // through and this timer can be ignored
        guard let request = weakRequest.value else {
            return
        }
        
        self.serviceRequests.removeFirst()
        
        request.callback(.failure(SBError.operationTimedOut(operation: .discoverServices)))
        
        self.runServiceRequest()
    }
`

jordanebelanger, 2023-04-13:
I see, any chance you also tried on more than one iOS 16.4 device?

Neha-Zco, 2023-04-13:
Yes, we have been experiencing issues on multiple BLE devices and phones running on 16.4 and later. But, it has also been noticed that on some iphones with 16.4 and later are not having any issues.

msn444, 2023-04-17:
I'm seeing similar on iOS 16.0 (iPhone 11 Pro). Intermittent -- sometimes it works and finds the services, other times returns nothing and times out. Test procedure is simply scanning for peripherals with a specified service, connecting to one, then calling discoverServices(withUUIDs:nil), then calling disconnect(). Often (but not always) this will be successful immediately after app launch, then subsequent runs of this sequence will fail to return services. As a test, implemented the same procedure using CoreBluetooth in the same app, and it successfully finds the services every time, even when SwiftyBluetooth is in this bad state where no services get returned.

jordanebelanger, 2023-04-17:
> I'm seeing similar on iOS 16.0 (iPhone 11 Pro). Intermittent -- sometimes it works and finds the services, other times returns nothing and times out. Test procedure is simply scanning for [...]

You can look at the code of the library internally if you want, nothing special/different is happening over simply calling `discoverService` by yourself.

maniknaidu, 2023-04-18:
Hello @jordanebelanger,

I have updated my device with iOS 16.4 I am not able to discover services and getting below error message. 
In below 16.4 devices while reconnecting its working fine, in above device its not working.
**Error Message :Bluetooth operation timed out: Discover services**

maniknaidu, 2023-04-18:
> I'm experiencing the same issue with iOS 16.4 and later versions.
> Have you been able to find a solution for this issue?

Did you find any solution?

jordanebelanger, 2023-04-20:
Ok I found the issue, the scan behaviour has changed on iOS 16.4 whereas you can now discover the same peripheral multiple times and each time the actual core bluetooth peripheral found is actually the same object instance. This led to some peripheral having their delegate overwritten by the duplicate Peripheral instance that were being found.

I will publish a new version soon.

jordanebelanger, 2023-04-20:
fixed in 3.1.0

closed 2023-04-20 by jordanebelanger